MaslowCNC/Mechanics #7: Motor_Mount.svg file seems wrong.

blurfl, 2017-03-15:
MakerCam and InkScape don't seem to find an outline in the file. Was it created from the appropriate view?

blurfl, 2017-03-15:
Here's a file that looks OK in MakerCam and InkScape. 
[Motor_Mount.svg.zip](https://github.com/MaslowCNC/Mechanics/files/846039/Motor_Mount.svg.zip)

mnelson78, 2017-03-15:
I opened it in Illustrator and it looks good now.

BarbourSmith, 2017-03-15:
Does PR #8 look like it fixed the issue?

blurfl, 2017-03-15:
MakerCam is happy with them all. InkScape sees the stroke as 90px wide, so renders them with hugely fat lines :( 
Should there be one for the sled, as well?

closed 2017-03-15 by blurfl

davidelang, 2017-03-16:
when doing CAD work in .svg (inkscape and others), you should always use an 
extremely thin line, otherwise you run the risk of the machine trying to cut 
both sides of the line.

On Wed, 15 Mar 2017, Scott Smith wrote:

> MakerCam is happy with them all. InkScape sees the stroke as 90px wide, so renders them with hugely fat lines :(
> Should there be one for the sled, as well?

BarbourSmith, 2017-03-16:
@davidlang I agree. The line exporter in sketchup doesn't seem happy exporting a line thinner than .05 in which isn't exactly narrow, but it is better than it was. I think I may need to download inkscape tomorrow to make them truly hair thin.

I've also added a file for the sled.

Do we feel good about merging PR #8 as is for now?

blurfl, 2017-03-16:
Bar,
 I’m using the .SVG exporter mentioned to Hannah by RifleSpiral , 20dThu Feb 23 2017 05:14:46 <http://www.maslowcnc.com/forums/#!/no-judgement:sketchup-is-friend-not-foe/theres-a-branch-on-github> and it is willing to do 3.54px (on the Mac I’m using…). No special setup, just using the 1mm defaults for line size.

blurfl, 2017-03-16:
Bar,
The motor_mount.svg file was not updated with the rest of them, it is still the fat one. Other than that, ready to merge.

davidelang, 2017-03-16:
It sounds like the right thing to do, so no reason not to merge the fixes

David Lang

k

BarbourSmith, 2017-03-16:
Thanks for the advice everyone. I'll make everything as close to zero width as I can first thing tomorrow. Yall are also welcome to make pull requests for this kind of thing if it's faster