MaslowCNC/Mechanics #20: Reduce friction between sled and cutting surface

BarbourSmith, 2017-03-22:
I suggested in [this video](https://www.youtube.com/watch?v=5pmNkcbVRkE) that we need to reduce the friction between the sled surface and the wood being cut. 

In the video I suggested that we could reduce the friction by pocketing out some of the area under the sled. It has been pointed out (very correctly) that friction has more to do with the weight of the sled, than it does with the surface area so creating a pocket can't actually reduce the friction. I think there is still some room to play with the idea of pockets under the sled for either improved dust collection or to distribute the weight differently because drag toward the edges of the sled produces greater torque to twist the sled than drag near the center.

The two factors which affect friction are the coefficient of friction and the weight of the sled.

We can reduce the coefficient of friction by using a slippery material like nylon or HDPE to make our sleds. I think that's a fantastic idea, but since it's not available to everyone I'd like to look for other solutions in the meantime. 

Reducing the weight of the sled is a good option. We don't really need to reduce the weight, only the force excreted by the sled on the material being cut, which we can control by adjusting the tilt of the machine. The current angle was decided back when we were using a much smaller less stable sled. I think our new larger sled could handle a much lower angle and still be stable.

davidelang, 2017-03-22:
I've been wondering about the angle, I figured I'd ask the beta testers to do 
some experimentation with steeper angles once we started getting several of them 
online.

That should let the sled be lighter as well (while keeping the same tension on 
the chains), which would also open up the possibility of faster movements for 
the design (either via motors with different gearing or larger sprockets)

mikberge, 2017-03-25:
first thing it come my mind, something cheap and easy to find....one cutting board! made with plastic similar to nylon or teflon .made to be durable and glide well. i think about two option , same form as the sled or many pocket under the sled plywood to insert many disc of cutting board strategicly these kind of plastic can be cut with the maslow

mikberge, 2017-03-25:
https://www.walmart.ca/en/ip/mainstays-pp-cutting-board/6000196138411

davidelang, 2017-03-25:
or this one that was posted a while ago

https://www.samsclub.com/sams/bakers-chefs-commercial-cutting-board-15-x-20/126035.ip?xid=plp:product:1:1

k

Seventhram, 2017-04-02:
Hi, I'm a beta tester.  I'm in the process of building my frame. I was a little concerned about the angle as well. I built my frame to be adjustable (using bolts on the top and bottom). Currently I have mine built to a 15 deg angle. I like the cutting board as a sled idea. I purchased a walmart cutting board, its rectangular, 20"x15"x3/8"thk. I'll most likely cut it to be more of an elliptical shape. Pic's to follow.

BarbourSmith, 2017-04-02:
Let us know what you learn!

buzz4209, 2017-04-03:
Many woodworking projects use melamine coated plywood or OSB for reducing friction. It's cheap and readily available.  Chamfering  the edges should be done too, as sharp corners may have a tendency to dig in and catch on any roughness present in your materials.

The cutting board idea is a good one, but I'm not sure the 1/2 inch thick polypropylene ones will work out long term. I'd worry about rigidity and durability based on the ones I have in my kitchen. :-) Best would be Corian (countertop type material) it's harder and slicker. Polypropylene will wear (scratch) much more readily than corian or melamine. It's generally a softer material. Unfortunately the Corian cutting boards are harder to find. I think Menards in the US often has them.

rb0087, 2017-04-08:
I think that ball transfer bearings would be the ideal solution to this problem, but it would potentially add a bit to the cost. Still, I think that it's an idea worth considering, because it would eliminate the issue of sliding friction with the wood. Instead, you would have rolling friction with the wood, and mostly rolling friction between steel ball bearings inside. The issue of it catching on the edges would also be largely eliminated. I think that you could source suitable ones for around $2 each in bulk, and if the design used ~10 of them, it wouldn't be a huge expense. It's also quite possible that some creative minds could think of a cheaper alternative that's inspired by this. Other than cost, the complication with this idea is that you would need enough of these to make sure that previous cuts in the plywood don't cause problems. If you had a perfect surface, 3 would be enough, but it would obviously cause problems if any of the 3 were going over dips in the plywood.

With the current design, regardless of what material the base is made out of, or what shape you have, with a flat base bearing on the surface, you'll still have sliding friction, and it will be difficult to predict because one of the materials is obviously plywood. Moisture content of the wood, the species/grade of the wood, etc. will all have an effect. When you have wood-wood, and wood on many plastics, you'll also see stick-slip behavior, which will make the motion difficult to control smoothly. This paper has some great info on the subject: [http://catalog.lib.kyushu-u.ac.jp/handle/2324/23785/p147.pdf](url) .

Also, if you do continue down the route of using a solid base like you currently have, I'd just like to point out that the shape does matter, regardless of the fact that you won't see the surface area anywhere in a simple friction equation. If this were lying flat on a table, and the forces were being applied right at the base, it might not, but in this case, the reaction forces at different points on the base will be different depending on how hard the chains are pulling (which will be applying a moment because the forces aren't directly in-plane), where the CG is, what the cutting force is and at what depth, what the grain direction is and what local imperfections are present, etc.  With uneven reaction forces comes uneven friction, which could be higher than a simple F=mu*m*g*sin(theta) would give you, and could be causing the base to rotate about its' center of gravity, which would cause positional errors at the cutting bit if it isn't the center of gravity.  Anyway, my main point is that it's not a simple situation, so I would trust your intuition that the shape is important. I think there's probably a bit of performance that could be gained by optimizing the shape.

BarbourSmith, 2017-04-08:
Thank you for that clearly well thought out and on point answer. I agree with everything you said. 

The idea that the shape matters is I think an especially important one. The shape is a variable that we have a lot of flexibility to play with. We have the ability to play with not only the perimeter shape of the sled, but also the contact surface area by creating pockets.

The paper is quite insightful. Thank you for linking that. I thought that this graph which shows the "stick-slip" behavior of different plastic types was particularly interesting:
[image]

I agree that transfer ball bearings would give the lowest coefficient of friction regardless of the cutting surface. Something I want to be careful of is having the price creep up. So many of the other CNC router projects (like the Shapeoko) have started in our price range, but have ended in the $1,000-$2,000 range as they add features and improve the design. The later models are undoubtedly better machines, but they just aren't affordable any more (at least to me). I think a good solution is to try to keep the bare bones machine cost down, and make more parts available as options. 

I think the final question is where we draw the "good enough" line. I am confident that we can reduce the stick behavior significantly by adjusting the machine tilt, sled shape, and sled surface material, hopefully to the point that it is not the dominant source of error. 

Again, thank you for that well thought out and on point comment and the helpful paper. I have a better understanding of the problem after having read it.

rb0087, 2017-04-08:
Glad you found it useful. I agree that the cost of the transfer ball bearings might be cost prohibitive, but I thought I'd throw that out there at least to put a bound on what the ideal no-compromises solution might look like. Even that idea does have its challenges, though, like dealing with wood dust contamination in the bearings. 

I also had another idea which would be much more economical, and would have many of the same benefits. If you had a pattern of carriage bolts going through the bottom of the sled, you would then have point contact with round steel surfaces on the wood, which I think would largely eliminate the stick-slip behavior and catching on edges. The coefficient of friction of steel on wood is roughly 1/3rd of wood-wood, static and dynamic coefficients of friction for steel-wood are almost the same (good for stop-start performance), and I think that point contact would make things more consistent. With zinc plated steel hardware, the added cost to the BOM would be around $2-$3 depending on size and number of fasteners. Stainless steel would at least double that. Here's a side/top view of what I had in mind:
[image]

As far as disadvantages go, it would be a bit of a pain to drill a bunch of holes for the fasteners and tighten them all down, and it would look a bit kludgy, but other than that, I think that it would be a cheap and relatively simple modification.  I wish I had one of these to try it out with!

davidelang, 2017-04-09:
well, drilling holes is not that hard, with an automatic Z it's trivial, but 
even without it, you stop and tell the person to run the router all the way down 
to drill the hole.

we'll have to wait and see what happens when someone tries this.

On Sat, 8 Apr 2017, Ray wrote:

> As far as disadvantages go, it would be a bit of a pain to drill a bunch of 
> holes for the fasteners and tighten them all down, and it would look a bit 
> kludgy, but other than that, I think that it would be a cheap and relatively 
> simple modification.  I wish I had one of these to try it out with!

Seventhram, 2017-04-09:
Hello everyone, I tried the cutting board idea. Nice idea but didn't work very well. 3/8" thk cutting board is too flimsy. See attached pic's. Pic #1 & 2 shows it being marked and mounted. Pic #3 shows the 1/2" gap the weight of the router and bricks cause. Plus I have the chain mounts too far apart. Pic #4 shows where I added a small piece of angle and bolted it together to make it stronger. Still didn't work. Today I'll attach the cutting board to some plywood.
I'm worried about how thick this becomes, too thick and the bits will be under too much stress.
[image]
[image]
[image]
[image]

Seventhram, 2017-04-09:
I added one more pic. This is what the framing and sled looks like.  Still work in progress
[image]

BarbourSmith, 2017-04-09:
@Seventhram It may not have worked exactly as you hoped, but it looks like a well done build. I would be interested to see what the effects of having the chains mounted that far apart are. My instinct says that spacing will have an effect on the tension in the slack chain when cutting over at the edges (which is the real goal of reducing friction), but I'm not sure what the effect will be.

@rb0087 We might have a beta kit left if you want to join in the fun. I'm not 100% sure, you'd have to email Hannah (hannah@maslowcnc.com) to find out. She's the one with the final tallies, but if there's one left we'd love to have your help!

jbarchuk, 2017-04-09:
Regardless of flimsy, could you get any idea of whether the sled moved
easier on the wood?

> Pic #4 <https://github.com/MaslowCNC/Mechanics/pull/4> shows where I
added a small piece of angle and bolted it together to make it stronger.
Still didn't work.

I see a piece of wood and 4 screws, and 'something' along the top edge but
I see the 'angle' thing. Yes, a piece of 1" aluminum angle should
strengthen the nylon drastically.

> Today I'll attach the cutting board to some plywood. I'm worried about
how thick this becomes, too thick and the bits will be under too much
stress.

Meaning wood on top of the cutting board? Yeah thickness is a concern.
However you can cut a large hole in the plywood such that the router goes
completely through it and mounts directly to the cutting board. The plywood
attaches only to the cutting board and has nothing to do with the router.
Yes it loses a little strength that you're trying to add but in % not very
much.

Seventhram, 2017-04-16:
Sorry for the delay (been working 6 day weeks).
I attached a piece of 3/8" plywood to the cutting board for strength and changed the shape.
Overkill on the amount of screws I used, I'm building long term strength.
I beveled the cutting board edge to assure it slides easier.
Last pic is fully assembled.
[image]
[image]
[image]

BarbourSmith, 2017-04-16:
That looks beautiful! Great work, and thank you for keeping us up to date. 

What are your feelings about the slickness factor? Can you feel the difference?

Seventhram, 2017-04-16:
Totally manual movement only, I have not connected any of the electronics. The sled slides across the wood with little effort.
Next is installing all the electronics, then calibrating, then testing.

blsteinhauer88, 2017-04-17:
Nice, could you mount the router directly to the cutting board,  I find the farther you take the router from the cutting surface, the Zaxis is running at the bottom the adjustment.  Or use the 1/4 inch Iron Wood avail at Home Depot for US users.  I put the furniture skids on mine and lifted the actual sled surface away from the cutting surface about 1/8 to 1/4 inch.  It is working good, but had a few problems getting thru the wood.  The router will only go so deep overall.  The further we lift away the router, the less depth it will travel.  
[image]
[image]

blsteinhauer88, 2017-04-17:
I just saw I repeated @jbarchuk 's advise in the end of his comment... Great minds....

jbarchuk, 2017-04-18:
blsteinhauer88 commented 
> Or use the 1/4 inch Iron Wood avail at Home Depot for US users.

I looked for that and found only flooring labeled 'finish' not even laminate. If you've seen a link for something I'd appreciate a copy of that.

I have a little experience with ironwood because there was some in the woods in the northeast region (and maybe elsewhere too.) I had a walking stick with about 15 rings and was only a little over an inch in diameter. And aside from not being much of it it would eat woodworking tools like jelly beans. Around here the most common furniture wood is ash. The thickest tree I ever saw was only about 6". The point being that it's not commercially viable for much other than decorative carvings and such.

blsteinhauer88, 2017-04-18:
Yea I searched Home Depot for the stuff.  It is plywood underlayment with a finished side.  It is not actually Ironwood, it think that is the brand.  I found the same thing on the Lowes site called revolutionply: https://www.lowes.com/pd/RevolutionPly-5mm-Poplar-Plywood-Application-as-4-x-8/50121135

jbarchuk, 2017-04-18:
> Plywood-Application-as-4-x-8/50121135

OK now I get it it's a brand. No problem. I just wanted to make sure ppl
understood what was being discussed and what they're BUYING with their HARD
EARNED CASH MONEY. I'm VERY concerned with the typical Chinese marketing
bait 'n' switch. It's RAMPANT. They have NOOOO problem with absolutely
literally LYING about product titles and descriptions. At Lowes, the 3/4"
plywood, they have the good stuff, and then for $2 less per sheet a Chinese
knockoff called Blondewood Birch. I asked the wood guy (who knows his stuff
because he was a long time carpenter (put out of work by the housing
crash)) if it was birch, and/or what blondewood meant. He hesitated and
hedged. I wasn't sure if he really didn't know, or if because of his job he
wasn't allowed to say. Regardless, it's 'mystery wood.' It's probably OK
stuff, though for the flex I saw over the full sheet, for $2 more I'd
really rather get something that I -know- what it is.

servant74, 2017-06-22:
From other groups, the two major problems with the Chinese plywood is the glue isn't consistent, and there are lots of voids internally (think of voids on some of the internal layers).

If I remember right, the PureBond brand at Home Depot is manufactured in North America (USA or Canada).  Also Plytanium at Lowe's is supposed to be a North American manufactured by GP (Georgia Pacific, but that may not be an official company name anymore.).

mrfugu, 2017-07-13:
I propose the cheapest, simple and effective solution to the sled friction problem is to cover the bottom of the sled with clear plastic shipping tape. 

Overlap strips of tape in one direction by a small amount (3-4mm) to ensure complete coverage, start with 1/2 the 1st strip of tape overhanging the top of the sled. 

Leave 1" of overhang all around and smooth on to the round over outside of the sled.  

Cut the router hole like a pie and fold up into the rounded over inside (router) hole. 

trim to make neat. 

replace/repair with new tape as necessary. 

This maintains the strength and workability of the plywood sled, its cheap, its accessible, its replaceable after wear, and is a lower friction option than bare plywood.

davidelang, 2017-07-13:
please test and report the results.
k

cswartzell, 2017-07-14:
I've created an extremely similar sled to seventhram, I'm lucky enough to have some 3/4" UHMW laying around the shop to make a decent circular sled. I'll post photos and a review of how it works shortly, but as I made this sled without bothering with a plywood temporary I don't really have anything to compare it to.

davidelang, 2017-07-14:
please make a plywood sled the same dimensions as your UHMW sled

the differences are most likely going to be seen when moving horizontally out to 
the edge on the bottom corners.

I did a test file that cut L shapes in the corner, dropping down vertically, and 
then doing the horizontal cut, and then going back the other direction. When the 
sled sticks, instead of getting a sharp L back, the corner of the L gets rounded 
as the sled has not moved as intended before it starts getting pulled up.

Experiment with the different sleds, and also with different weights and angles 
of the machine.

TheRiflesSpiral, 2017-07-14:
Just a note of caution with UHMW; it's stretchy. It's also not dimensionally stable when machining so if you're drilling holes/surfacing/counter-sinking, etc, be prepared to deal with some warpage.

In fact, I wouldn't advocate any plastic-only sled; a stable base is needed and some contact cement should "permanently" bond the two.

cswartzell, 2017-07-14:
I was definitely going to mount the Chain Brackets to an +1/8" metal plate, and mount that to the sled, I was worried that the force of opposing chains near the top of the machine could stress the plastic. If I take that step, then mounting holes for the router and weights aren't concerning to me, its not like I'm tapping the sheet. If the stress of chains is resisted by a metal support, then there is little force to stretch the sled to any significant degree.

 Then again, attatching the sheet to a backer would mean I could mount the hardware to said backer and thus have almost no holes in the UMHW itself, which are likely to get filled with dust and add to friction. Considered bonding PTFE to a wood sled, but chose UMHW for vastly superior abrasion resistance. If you only cut smooth things (MDF is probably 90% of what I'll cut anyway), then oiled PTFE might be the ultimate friction reducer. Dynmaic Coeficion of Friction under 0.03, so slippy! https://www.redwoodplastics.com/wp-content/uploads/2010/08/PTFE-Properties-metric.pdf

TheRiflesSpiral, 2017-07-14:
That's an interesting thought. Do all your counter-sink work in the stabilizer then glue the slip agent over the top... That's certainly the lowest friction/least chance for hanging up configuration I've considered. As long as you have a reliable way to secure your fasteners (as studs) and never want to change it, that's probably as good as it gets.

gator141, 2017-07-17:
Has anyone tried a spray silicon or teflon to reduce the sliding friction on the sled?  I just received my kit and am starting to build and that seems to be a quick method to reduce sliding friction

davidelang, 2017-07-17:
not that anyone has reported yet.
k

msmollin, 2017-08-06:
How are the UHMW sleds working out for people? Finally getting around to setting up my machine and am staring at the 1/2" stuff on amazon - thinking of doing that  with a 1/4" birch plywood as a backer to keep everything stable. I would do 1/4" UHMW but apparently it's warped pretty badly coming from amazon.

Thoughts?

TheRiflesSpiral, 2017-08-07:
You would be better off using a thin layer of UHMW and a thick layer of plywood. You want the rigid structure to be as stiff as possible, but the slippy layer can be as thin as is reasonable to work with. I haven't tried it yet but I'm fairly certain that some contact cement would permanently bond the two surfaces with no problem.

msmollin, 2017-08-07:
Agreed on the contact cement. I hung a 3/32's sheet of magnetic stainless weighing about 40 pounds on a wood wall using standard contact cement you buy at home depot. Has stayed up for a year with us loading it with magnetic knife racks and a spice rack without an issue. Plenty of surface area though (~2ft x ~5ft) for it to grab onto though.

Yeah I need to find a better source for the UHMW I guess since Amazon reviews say the 1/4" stuff is badly warped coming from them. Every other place I've seen online though is rather expensive for shipping. It's about 45 minutes drive to the nearest place that may have it from me, although I am unsure they even sell to the general public (making calls today).

skiz, 2017-11-14:
For slickness, laminate flooring is cheap, or some self stick vinyl sheet would be ideal.
Ball bearing casters can be had quite cheaply (~$1 on amazon) and come in nylon and steel. They can be inset from the bottom to minimize height.

Add a few coats of urethane to the bottom of your sled?

An air hockey table sled would be cool? (reverse your vacuum, it doesn't work that well anyway :P).

Remove the weights, add a bungee to the bottom center of the frame to the sled, and tilt the machine more.  It should provide comparable stability for cutting, but a better coefficient for movement.

davidelang, 2017-11-14:
On Tue, 14 Nov 2017, Josh Martin wrote:

> For slickness, laminate flooring is cheap, or some self stick vinyl sheet would be ideal.

or UHMW plastic (available as cutting boards cheaply)

> Ball bearing casters can be had quite cheaply (~$1 on amazon) and come in nylon and steel. They can be inset from the bottom to minimize height.

they will have a tendency to fall in to areas you've already cut.

> Add a few coats of urethane to the bottom of your sled?

yes

> An air hockey table sled would be cool? (reverse your vacuum, it doesn't work that well anyway :P).

would be very cool.

> Remove the weights, add a bungee to the bottom center of the frame to the sled, and tilt the machine more.  It should provide comparable stability for cutting, but a better coefficient for movement.

see the discussion about adding a third motor that we had a few days ago, it 
turns out this isn't nearly as easy as it sounds at first. When you are cutting 
near the bottom corners, the bungee cord would be pulling in the wrong 
direction, making a bad problem worse.

matthewkleinmann, 2017-11-15:
Some thoughts.  I don't have a router yet, but I have built a lot of things..

1) The ball bearings.  I am not a fan of this idea because as the piece the sled is gliding over starts to get tracks touted out of it, I would fear that unless the sled is perfectly balanced one or more of the bearings might get stuck or skip as it attempts to cross the void.

2) The cutting board.  I like the concept, but perhaps try making the sled out of aircraft grade plywood or even aluminum.  Yes, it is more expensive.

3) I am looking forward to hearing how the packing tape worked out.  Thin and inexpensive.  Perhaps that over aluminum as I think it would adhere better to a totally flat surface.

4) I worked on an experiment once and we bought thin adhesive backed teflon.  I don't recall the source or the price.  I suspect it was not inexpensive, but it was a really good material. and this use would only require a small amount of it.  Again, over wood or aluminum.

5) My last thought, more out of the box and harder to effect would be to make the sled something like an air hockey table powered by a shop vac.  This would give you some degree of weirdness in the Z axis, but I suspect for the vast bulk of the projects people are building with this router, that is acceptable.  For the things I would like to build, the Z axis is pretty much in or out.  FWIW, I got enamored with the air hockey idea pondering building extension feed tables for my planer.

davidelang, 2017-11-15:
On Wed, 15 Nov 2017, matthewkleinmann wrote:

> 5) My last thought, more out of the box and harder to effect would be to make
> the sled something like an air hockey table powered by a shop vac.  This would
> give you some degree of weirdness in [...]

This could also have problems going over earlier cuts (just like the bearings 
approach)

servant74, 2017-11-15:
Another approach, is to use teflon.  Contact your local plastics supplier,
and get some thin Teflon or PTFE sheets that have been acid etched.
This will allow epoxy to attach well and stay.  Mix your favorite epoxy and
cover the sled bottom, and put on the sheet (etched side toward the epoxy)
and leave under some amount of compression (to keep things from springing
apart while the epoxy cures).
The sled can be cut/trimmed with normal tools to make way for smooth edges
etc.  Teflon is really pretty soft.

This was used when my Dad worked for Bell Helicopter on some of the parts.
They used high temp epoxy and put it in clamps then in an autoclave under
pressure.  But hey, it was for aerospace use, so money was no object.  We
used their cutoffs/scraps of Teflon like this to make furniture slides for
use around our home.  It worked well.

From a quick look online, the acid etched PTFE that is thin (say .02
inches) runs about $1.80/sq ft USD.  There may be minimum orders plus
shipping, but such is life ordering industrial goods.  Hopefully you can
find some without to much overhead involved.

davidelang, 2017-11-15:
On Wed, 15 Nov 2017, Jack in TN wrote:

> Teflon is really pretty soft.

I would worry that the Teflon will get chewed up by the edges of the wood and 
chips (note that I worry about a lot of things)

matthewkleinmann, 2017-11-15:
David: On the air levitated approach, I had the same thought about the air levitated sled, but if you look at an air hockey table, most of the holes are uncovered most of the time.  For me the show stopper there is how complicated it is.

On the teflon, the stuff we had was tough.  It was akin to the stuff they use in dryer bearings.  Not quite that heavy but definitely tough stuff.

blurfl, 2018-01-10:
This is a valuable discussion, but not really an issue. Is there a way to move this to the forum or the wiki?

BarbourSmith, 2018-01-10:
I am not sure how to move it to the forum other than a copy and paste 😐

davidelang, 2018-01-10:
and how should we tag things like this? "discussion"?

On Wed, 10 Jan 2018, Scott Smith wrote:

> Date: Wed, 10 Jan 2018 19:09:04 +0000 (UTC)
> From: Scott Smith <notifications@github.com>
> Reply-To: MaslowCNC/Mechanics
>     <reply+0021b24f6c5df6b2860e6de4fc51de1c10632af5c85f27f192cf00000001166e26d
>     092a169ce0ce20054@reply.github.com>
> To: MaslowCNC/Mechanics <Mechanics@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [MaslowCNC/Mechanics] Reduce friction between sled and cutting
>     surface (#20)
> 
> This is a valuable discussion, but not really an issue. Is there a way to move this to the forum or the wiki?
>
>

steenblikrs, 2018-03-24:
@servant74 > Another approach, is to use teflon.  Contact your local plastics supplier,
and get some thin Teflon or PTFE sheets that have been acid etched.
This will allow epoxy to attach well and stay.  Mix your favorite epoxy and
cover the sled bottom, and put on the sheet (etched side toward the epoxy)
and leave under some amount of compression (to keep things from springing
apart while the epoxy cures).
The sled can be cut/trimmed with normal tools to make way for smooth edges
etc.  Teflon is really pretty soft.

Did anyone actually try a PTFE sled?? I'd love to hear successes failures!

I found some discussions (links below), that I guess, is a continuation of the above conversation? It seems a little confusing to have discussions on the Github and in the Maslow forum.. I'm not sure how to correct that, but just my 2¢.

https://forums.maslowcnc.com/t/polycarbonate-sled/2792/15
https://forums.maslowcnc.com/t/possible-table-round-sled/2780/6

dundermane, 2018-04-05:
I've been toying around with the idea of using the blower side of my shop vac to hover the sled.  Just like an air hockey table.

matthewkleinmann, 2018-04-05:
> A lot of people have talked about doing this, but nobody has tried and
> reported
> on it
>
> issues to worry about:
>
> as you go over existing cuts, the pressure under the sled can change, and
> that
> will cause the height that the sled 'floats' over the workpiece to change,
> affecting your cut depth.

>issues to worry about:

>as you go over existing cuts, the pressure under the sled can change, and
that
>will cause the height that the sled 'floats' over the workpiece to change,
>affecting your cut depth.

True, but for 90% of the people using these the router is going to be
either through the piece or not cutting at all so that fraction of an inch
does not really matter.

Circus621, 2018-06-30:
Im not sure if this moved or has been closed but just use melamine and round over the edges. Works great, reduces friction and is cheap and available. Why make it so complicated?

BarbourSmith, 2018-07-02:
We've seen so many great solutions to this that I am going to close it

closed 2018-07-02 by BarbourSmith