MaslowCNC/Mechanics #24: Building temp frame step 10

sanenerd, 2017-03-25:
docs say
Step 10: Attach Cup Hook to top of Plywood Sheet

Attach 1 Cup Hook (found in the Small Hardware Bag) to the top left corner of the sheet of plywood below and slightly to the right of the motor shaft.

Pic show attaching to 2x4.. I assume photo is right 

PS. I am sorry if I am putting this in the wrong place

BarbourSmith, 2017-03-25:
Great catch!

This is absolutely the right place!

The picture is correct, I'll change the text right now (you also have the ability to edit it directly using the "Edit" button in the right corner if you want)

BarbourSmith, 2017-03-29:
I'm going to close this if that's ok. I think it should be up to date. Thank you again for pointing that out!

closed 2017-03-29 by BarbourSmith