MaslowCNC/Mechanics #29: Mechanics Sketchup files cannot be opened

jonasdagefoerde, 2017-04-06:
The Mechanics/Machine.skp and Mechanics/TempSeld.skp cannot be opened with Sketchup (latest version: 17.2.2554) on OSX (10.11.6)

BarbourSmith, 2017-04-06:
This is a pretty common problem and I'm not sure what the solution is. I've heard from people that if you use the download button and download the zip file then they work, but if you download each file individually they don't.
[image]

We need to find a better way so we stop confusing everyone, I'm not sure what the better way is.

jonasdagefoerde, 2017-04-06:
Hi bar,

thanks for the tip: now it works :) May be you can add a hint to the README file for the time being - until you found a solution.

Cheers
Jonas

BarbourSmith, 2017-04-06:
Great suggestion!

closed 2017-04-06 by BarbourSmith