MaslowCNC/Mechanics #30: NOT an issue, just need to host image file of alternative frame design

aluminumwelder, 2017-04-13:
this is NOT an issue, just need to host image file for new wiki page I made, couldn't figure out any other wayt to do this. 
[image]

BarbourSmith, 2017-04-13:
They way I have been hosting pictures is to put them in the "documentation" folder on GitHub, there's a guide for how to do that: http://www.maslowcnc.com/howtocontribute/

But if you don't want to do that whole process I can put them in there for you.

aluminumwelder, 2017-04-13:
ok sorry

BarbourSmith, 2017-04-14:
Not to worry, good solution. I'm going to move the picture and I'll just update where the link points on the wiki page. Thank you for making a wiki page!

BarbourSmith, 2017-04-14:
Great, I've added the photo to the repo. Awesome work!

closed 2017-04-14 by BarbourSmith