MaslowCNC/Mechanics #31: Unclear router description and features

jbarchuk, 2017-04-14:
Re; https://github.com/MaslowCNC/Mechanics/wiki/More-Photos
Bar, your page with the Bosch router explains a lot but might be confusing to users. First, it doesn't include a specific model number. I managed to track it down by the nameplate pic but there might still be oddities. These manufacturers build tools with different features and varying part numbers, and probably different features/numbers for different world regions.
Second, more importantly, is that the first pic might be misleading. That router is AAMOF the only one I've seen that has 'fixed base adjustable height' AND plunge style 'travel stops' features. It does that by giving the adjustment rod a latch mechanism that slips into the rotating 'stops wheel.' A user might see that wheel, see it on another router, and not understand the lack of the notch at the bottom of the adjustment rod, and get a router that can't do what's needed.
I think those pics should be moved to the 'router reviews' page with an explanation that it's a 'dual style' router suitable for Maslow's needs but unusual/different from other routers.

GeroBH, 2017-04-15:
@jbarchuk All pictures was originally here:   https://github.com/MaslowCNC/Mechanics/wiki/Tested-Routers-List
I admit, too many pics. Will try to make that more compact. They were moved to 'More-Photos' to make the router list better to overview.
Behind the model number there, it is mentioned that this router is not recommended and not tested. The 'not tested' will be removed today, because it is moving and it will cut. The 'not recommended' will remain, as this is one heavy beast (7kg without bricks and sled), to high, bringing the centre of gravity far over the sled. Moving it to the top edge of the cutting sheet made one of my motor mounts bend and a chain jump over the sprockets. It is sure not in the line of the goal 'Make it possible for everyone to have a Maslow', more on the line of testing how far you can go to reach the limits of the design. May I add more pictures of how the Z-Axis is mounted?

BarbourSmith, 2017-04-16:
I think more pictures clarifying how the z-axis is mounted would be a great thing. I think @jbarchuk is right that we should make it clear that it's a unique router type. @jbarchuk do you think it would be possible to make it clear without adding more pictures, just to keep the router review section clean? Hopefully we will soon have a number of routers listed there and I'm worried about it becoming crowded.

jbarchuk, 2017-04-17:
I didn't mean to give the idea that there were 'too many' pics. Absolutely
not at all. The more the better especially around the z-adjustment
hardware! Beyond that all these routers are just a motor and base.On one
level, there are two kinds of users. The first user has a router and needs
to decide if it'll work for Maslow. The second kind has particular projects
and needs in mind, and needs to pick a router.
On another level there are similarly two kinds of users. One (me for
example) has a very narrow-specific task, and they need to pick the router
with the best features for that task. The other user has potentially many
different kinds of projects in mind, has no router at all, and needs to
pick one with a wide enough range of variables that could meet all of those
tasks. For those folks there's more compromise required because there will
be no 'perfect' router.
There's no way to describe, show, and test every router on every material
under every condition. The best a web page of pics and descriptions can do
is say 'it does [this] really well and [that] average to poor.' It's up to
the individual user to decide if what they have will do what they need, or
which one they need to get to do what they need.
Again (heeere we go again :) ) this is where m00t and wiki are -not- good
software for these kinds of displays, descriptions and discussions. Yes,
one huge long page that has every photo and detail of every router can do
it, it's just -easier- to read one page about one router and not have to
bother with the others.
-More- information can -never- be a bad thing!!
Heck the 'too big too heavy' thing that you experienced Gero might be a
deal -maker- for someone who wants to drive a 1" bit through 1" oak!
Whether the -frame- is a little too weak to handle the high end of the
power bell curve is a different issue, but it's exactly those kinds of
tests that that user wants to read about. It's -very- easy to make the
frame stronger if necessary, and the chains and motors are certainly
already strong enough. As has been seen the exact frame build style is not
cast in stone. The intent of the Maslow concept is not 'this design can do
any and every task,' but that it can be tweaked and recast to do many
different specific tasks.
A related topic, whether the style of g-code generation of a particular
software, and exact firmware operation is suitable for any task is up in
the air. It's possible that there will be different power ranges that one
style works best for lightweight stuff and a different style works better
for heavy stuff.
This is all very similar to how plain router (or any power or unpowered
tool) operation works in real life. It takes practice and experience to
make any tool perform a task. Folks want to read about specific experiences
to see if things will perform nicely for what -they- want to do. If a
router is out of the range of what a user needs they'll have to decide if
the tweaks will be enough to bring it in range whether it's shallower cuts
or slower cuts or a different kind of bit or even a different frame angle.

GeroBH, 2017-04-17:
Thank for the comments. I could add something like a watermark with model and type on the picture itself, like GOF 2000 CE / Fixed Base. What do you think?

BarbourSmith, 2017-04-17:
I think it looks great. I didn't realize how big that router is until seeing the z-axis attached. Wow, it's huge! Overall, I think that the changes look great. Thank you for the update.

mrfugu, 2017-07-15:
I gave each router their own wiki page linked from the 'tested routers' page, this issue can probably be closed.

closed 2017-07-17 by BarbourSmith