MaslowCNC/Mechanics #32: Front Arms point needs rounding

mnelson78, 2017-04-17:
The front arms file goes to a sharp point.  This is the point that sits on the top of the sheet of plywood and points horizontally.  When cut out in plywood the point just chips away.  This also creates a weak point that is easily broken during the building of the arms.  I suggest rounding this point, perhaps a 1" radius?

BarbourSmith, 2017-04-17:
Great suggestion and an easy change

mrfugu, 2017-06-21:
I'm happy to make this change to the Fusion360 file I've created (http://a360.co/2sLJASm). 

If you'd like, I can round that corner over and export out to a new Arm_Front.svg as well?

BarbourSmith, 2017-06-21:
@mrfugu, that would be great if you want to take that on! 👍 👍 

I've been working on a frame redesign that dramatically reduces the part count and I just updated the github with the new versions of all the files so be sure to work from the latest files so we don't waste your time.

Cool Fusion model!

davidelang, 2017-06-21:
It's really nice with an update simplifies construction :-)

mrfugu, 2017-06-21:
Hey thanks, glad to assist.  I'll update the fusion project with your new SVGs 1st, then make any changes needed. will update once it's live.

mrfugu, 2017-06-21:
I've created Pull Reqs with the updated versions of 'All_Parts_Laid_Out' and "NewArmShape' incorporating a 3mm fillet to round over the sharp points in the arm, and widened the Brick Holder holes just slightly (from 6 to 7 mm) to allow Fusion360, in particular, to easily program those holes with a standard 1/4" plunge bit.

mrfugu, 2017-07-13:
this issue can be closed.

BarbourSmith, 2017-07-13:
@mrfugu  Thank you for flagging all of these

closed 2017-07-13 by BarbourSmith

mrfugu, 2017-07-13:
no sweat. rock on!