MaslowCNC/Mechanics #51: Chain link height

ronbutterfield, 2017-05-20:
Please add a note to step 18, that the height the chain is attached to the sled should match the center of gravity of the router/sled assembly.  This may need to be adjusted for cutting very thick or thin materials.

davidelang, 2017-05-20:
and ideally you want the chain to be straight, so you want the motors out far 
enough for your 'standard' material (say an inch from the frame + what's needed 
for the center of gravity for the sled)

BarbourSmith, 2017-05-23:
That's a great point, it should be clearer.

Feel free to edit the wiki page directly when you come across something like that, or I can.

blurfl, 2018-01-10:
Has this been accomplished? If not, volunteers? 😁