MaslowCNC/Mechanics #52: Detailed specs for motors and encoders

rodrigogoncalves, 2017-05-28:
Hi, is the [BOM](https://github.com/MaslowCNC/Mechanics/blob/master/BOM.txt) the official specs of the parts used? I'd like to know the makes and specs for the motors and encoders. Thanks

davidelang, 2017-05-28:
the motors are custom built, they have a 291:1 worm gear with a 7-line encoder 
on the motor, resulting in 8148 pulses/rev of the output shaft, they have a 
final drive torque of 30Kg/cm and turn at 20 rpm

Enjoying28, 2017-06-06:
I wonder if we can make a machine with these materials.

GM4632-370 DC 12V 30RPM High torque Turbo Encoder Motor Worm Geared Motor Reducer Motor
http://a.co/6l5dS2B
Browning 25B10 Minimum Bore Roller Chain Sprocket, Single Strand, Steel, 1/4" Stocked Bore, 10 Teeth
http://a.co/3dT9nYc
25 Roller Chain 50 Feet with 5 Connecting Links
http://a.co/0Qzel3k
Makeblock MegaPi - Born to Motion Control for Maker and Geek
http://a.co/iL2OnRn
Makeblock Megapi Encoder/DC Motor Driver 
http://a.co/3wgzsT6
NEWSTYLE 12v 30a Dc Universal Regulated Switching Power Supply 360w for CCTV, Radio, Computer Project
http://a.co/7Ea39eI

maybe would have to edit the step factor

davidelang, 2017-06-06:
looks plausable, I'll point out that you are driving three motors which take 
<0.5A each, so your 30A power supply is drastic overkill

The motor's rated torque is less than the maslow motors, but the worst that will 
happen is that this means you have to go slower.

there will probably be some software changes needed to make the software run on 
that board (and it's not clear from the Amazon page how much flash is on that 
board, so double check that there is enough space.

David Lang

On Tue, 6 Jun 2017, Enjoying28 wrote:

> Date: Tue, 06 Jun 2017 15:12:10 -0700
> From: Enjoying28 <notifications@github.com>
> Reply-To: MaslowCNC/Mechanics
>     <reply+0021b24f5bb304806bbd1e1ba6d837adc87c0d95f505747192cf00000001154eeab
>     a92a169ce0dd252c1@reply.github.com>
> To: MaslowCNC/Mechanics <Mechanics@noreply.github.com>
> Cc: David Lang <david@lang.hm>, Comment <comment@noreply.github.com>
> Subject: Re: [MaslowCNC/Mechanics] Detailed specs for motors and encoders
>     (#52)
> 
> I wonder if we can make a machine with these materials.
>
> GM4632-370 DC 12V 30RPM High torque Turbo Encoder Motor Worm Geared Motor Reducer Motor
> http://a.co/6l5dS2B
> Browning 25B10 Minimum Bore Roller Chain Sprocket, Single Strand, Steel, 1/4" Stocked Bore, 10 Teeth
> http://a.co/3dT9nYc
> 25 Roller Chain 50 Feet with 5 Connecting Links
> http://a.co/0Qzel3k
> Makeblock MegaPi - Born to Motion Control for Maker and Geek
> http://a.co/iL2OnRn
> Makeblock Megapi Encoder/DC Motor Driver
> http://a.co/3wgzsT6
> NEWSTYLE 12v 30a Dc Universal Regulated Switching Power Supply 360w for CCTV, Radio, Computer Project
> http://a.co/7Ea39eI
>
> maybe would have to edit the step factor
>
>

mrfugu, 2017-07-13:
an optional/equivalent parts list has been created. 
this issue can be closed.

closed 2017-07-13 by BarbourSmith