MaslowCNC/Mechanics #53: All_Parts_Laid_Out.svg AngleBrace1 error

mrfugu, 2017-06-08:
In the file "All_Parts_Laid_Out.svg" the patterns (2) for the part called 'AngleBrace1.svg' has a sub-mm inconsistency in its long sides. 

This doesn't seem to exist in the "AngleBrace1.svg' file itself, just the "All_Parts_Laid_Out.svg" file. It's minor, but probably adds unnecessary complexity to the tool paths generated for this part using this consolidated file. 

-mrfugu

mrfugu, 2017-06-08:
Correction: This error appears on the long 90 deg edge of the AngleBrace1 shape, which appears 4x in the " "All_Parts_Laid_Out.svg" file

BarbourSmith, 2017-06-08:
Good catch! Thanks for noticing that. If you want to make a pull request which fixes it, I would love to merge it. If not I'll look into correcting the issue.

mrfugu, 2017-06-08:
Cool, will do it now.

mrfugu, 2017-06-08:
this is my 1st Github contribution, apologies if I made a mess of it.

mrfugu, 2017-06-08:
Well, the file is in my post, but it reads to me like i requested a merge of recently changed files as well, not my intent.

BarbourSmith, 2017-06-08:
😀 You ARE helping!

It seems like maybe a bunch of other files got wrapped up in the pull request for some reason, maybe give signing [the guest book](http://www.maslowcnc.com/howtocontribute) a go to test the process out

mrfugu, 2017-06-08:
Hey, thanks! 

	Sorry about that, I’ll read more instructions before any future contributions, hope I didn’t make too big of a mess! 

	Cheers, 

Robert Crecine
429 1st Ave 
Suite 8
Pittsburgh, PA 15219
412-580-8618

BarbourSmith, 2017-06-08:
The beautiful thing about Git Hub is it's virtually impossible to mess anything up. Every change is logged and reversible so you can pretty much always go for it! 👍 👍

mrfugu, 2017-06-19:
Ok, I think I got it right this time, I re-submitted the corrected 'All_Parts_Laid_Out.svg' file.

BarbourSmith, 2017-06-19:
Just merged it. Fantastic work! Thank you. 
👍 👍 👍

closed 2017-06-19 by BarbourSmith