MaslowCNC/Mechanics #62: All parts SVG missing parts

ladyhawke13, 2017-06-28:
Shame on me for not going through the step of verifying - the All parts laid out SVG is missing the back piece of the arms, as well the two middle angle braces(the non-triangular ones). Also missing the SVG for those individual parts.  

For non-beta users a measured diagram with all plywood and 2x4 parts numbered would be very helpful(speaking as someone who used to help with flat pack assembled furniture instructions!). I'll be winging it with scraps and likely using my jigsaw for the back arm piece, as I'd started disassembling my temporary frame before realizing that I was missing things.

BarbourSmith, 2017-06-28:
You caught me right in the middle of re-writing the instructions to not need those parts. 

The new simplified design uses a 2x4 for the back of the arm part, and removes the middle brace from the legs to simplify construction.

I'll be re-writing the written instructions and taking pictures today.

ladyhawke13, 2017-06-28:
Leave it to me to pull at the wrong time! I have the frame supports together well enough, I just need to do the arms. I'll be retrieving my jigsaw from a different project site tomorrow and hopefully can get those cut out Friday.

BarbourSmith, 2017-06-28:
The new arms are mostly 2x4s, with the face plate mostly there for decorative reasons, You can really just use the 2x4s without bothering to cut the front parts if you want.

It will be more clear once I get the instructions written 😀 

[image]

ladyhawke13, 2017-06-29:
Ahh, gotcha! I definitely have more than enough 2x4 scraps floating around to do that. I have the front pieces of the arms, just not the back. So easily enough sorted :)

mrfugu, 2017-07-13:
this issue can be closed.

closed 2017-07-13 by BarbourSmith