MaslowCNC/Mechanics #79: Avoiding tearing down the temporary frame

skiz, 2017-11-14:
I was able to avoid removing most of the temporary frame (including all of the electronics, motors, etc) by taking a few precautionary steps along the way.  I have also avoided adding the additional arm facades, as I know my arms are both 20" out at 45 degrees.  I may revisit this at a later time, but it saved quite a bit of work and lots of recalibration time not having to deal with tearing everything apart.

1. Unplug anything that could move
1. Detach the sled (and z-axis if you have one), and use the pins to keep the chain attached on the gears (as close to the gear as possible).
1. Unhook the bungees from the tensioners, and safely remove and store the tensioners/hooks/gears.
1. unscrew the lower temporary legs as per the instructions
1. measure and screw the 2 6" blocks to the back/top of the frame.
1. build the permanent legs and mount per the instructions
1. simply hook back up the tensioners and remove the pins from the chain.

I figured this may save someone else some time, but being an "unofficial" hack, I'm adding it here.

BarbourSmith, 2017-11-14:
I am going to fully revisit the frame design and build process before batch three and I will try to incorporate these ideas, thanks for the tips!

blurfl, 2018-01-10:
Good suggestions, have they been incorporated? Is this issue needed as a 'to do' item?