MaslowCNC/Mechanics #85: Frame flex

BarbourSmith, 2018-01-15:
The stock frame has a measurable amount of flex in it which is becoming an important factor in calibration. 

What is the easiest way to reduce the amount of flex in an existing frame while changing it the least amount?

blurfl, 2018-01-15:
I suggest a 10' 2x4 screwed to the arms just below the motor mounts as inexpensive and needing the fewest tools.

BarbourSmith, 2018-01-15:
Here is my first proposed solution:

[image]

It adds a 2x4 between the two arms. It requires an additional 2x4 and that the arm braces be re-cut, but no other changes are needed

BarbourSmith, 2018-01-15:
From playing with my current frame, there is some play between the motor mounts (the plywood which the motor is attached to) and the 2x4 that it is attached to. Is there any way to reduce play there? It seems like having both of the motors bolt directly to a long 2x4 is the best solution but a little bit harder to implement.

blurfl, 2018-01-15:
With  The motor mounts angled that way, there  could still be flex at the top of the work area. Turning them horizontal would allow you to screw them to the 10 foot 2 x 4, is that what you’re suggesting?

BarbourSmith, 2018-01-16:
I agree that we could still get flex at the top of the work area this way. I'm thinking that what we really want to do is what @dlang did where he moved the entire bar that the motors are on forwards so that the motors attach directly to the 2x4. I haven't thought of a way to do that which I really like yet

davidelang, 2018-01-16:
add a 10' 2x4 across the top connecting the wings. If you have the plywood  arm 
covers, you can put the 2x4 in front of them and screw from the back.

This isn't the best new frame, but it's the quickest and easiest way to modify 
an existing frame.

davidelang, 2018-01-16:
The best new frame is going to be very different from the best modification to an existing frame

modifying an existing frame is simply adding a 2x4, no need to modify the plywood parts

[image]

blurfl, 2018-01-16:
Adding the 2x4 in this manner helps brace the angled motor mounts as well - double win!

davidelang, 2018-01-16:
a top beam design that is similar to the original frame is
[image]

davidelang, 2018-01-16:
But I really like this folding, wheeled frame concept (and note that this makes it easy to move the top  bar forward)

[image]

all part of the onshape document at https://cad.onshape.com/documents/e635c24e358635f51da4b399/w/5a63b67113542f248fbe6d7b/e/50fff2211306b5f2bfa5dd3c

davidelang, 2018-01-16:
If we were willing to call a unistrut top beam the standard, we could have motor mounts made that would wrap around the strut and had a cap over the ends so that they were easily removable (with one bolt to hold them tight)

otherwise, you can just screw them to a 2x4.

Remember that the brackets don't have to be super strong, they need to support the weight of the top beam (~11 pounds for a 2x4) the sled (~20 pounds), the chain (~2 pounds), so call it ~25 pounds. That's not much and just about anything can do that without noticeable flex.

I didn't want to be screwing into the end-grain of the wood, so I already was planning for short lengths of 2x4 for the corners, so I just extended them to 12" long.

blurfl, 2018-01-16:
> call a unistrut top beam the standard

Before we do that we need to check in with users in other parts of the world... Steel strut might be hard to find and expensive outside the major urban areas of the US.

davidelang, 2018-01-16:
@blurfl 

I agree, although unistrut is used for _so_ many things that it's probably available.

unfortunately, I can't figure out a good way to bend the material that would stay on the front of the material (so that it didn't care how deep it was)

blurfl, 2018-01-16:
@davidelang , I'm not sure I followed that...

davidelang, 2018-01-16:
2x4 is 1.5" thick, unistrut is 1  5/8" thick, pretty close to the same.

so if you have a motor mount that covered the bottom, end, front, and top of the beam, it would work with either unistrut or 2x material.

But I can't figure out something that's workable.

davidelang, 2018-01-20:
a note on the top beam attachment.

The alternate 1 design used a 2x10 on the top, with the idea that you can drill a series of holes in it and slide it forward an inch or two (a bit more if you use a 2x6 as the top board of the frame)

but a 2x10 is on the heavy side (heavier than a unistrut) and while this has some adjustability, it's not a lot.

So with the alternate 2 design, I extended the corner braces out so it doesn't depend on overlapping boards, which lets you use a unistrut or 2x4 for the top beam and move it to whatever distance you need. I made the supports 12", but it looks like 8-9" is about right for standard plywood.