MaslowCNC/Mechanics #86: Wiki page claims 0.25in router bit but links to 0.1875in router bit.

abetusk, 2018-01-18:
The Wiki page for the [0.25in 2 Flute, Tungsten, Up Spiral Bit](https://github.com/MaslowCNC/Mechanics/wiki/0.25in-2-Flute,-Tungsten,-Up-Spiral-Bit) links to a bit sold on Home Depot that is in actuality a [0.1875in cut diameter](https://www.homedepot.com/p/Diablo-1-4-in-Up-Spiral-Router-Bit-DR75101/204073552).  The **Min Chuck Size Required** is `1/4 in.` but the **Cut Diameter** is `0.1875 in`.

I assume this bit should actually be a 0.25in cut diameter?  Could you please link to the proper router bit?

BarbourSmith, 2018-01-18:
Great catch! I vote we don't recommend that router bit at all anymore because there is a comparable one in our store for half the price. Although maybe it's a good idea to link to one that is available locally to most people. What does everyone think?

davidelang, 2018-01-18:
On Thu, 18 Jan 2018, BarbourSmith wrote:

> Great catch! I vote we don't recommend that router bit at all anymore because 
> there is a comparable one in our store for half the price. Although maybe it's 
> a good idea to link to one that is available locally to most people. What does 
> everyone think?

include both (with price/date fields that can be updated by users)

just listing ones you sell can leave a bad taste in people's mouth, even if it 
is cheaper than the other one.

abetusk, 2018-01-19:
For people trying to build their own MaslowCNC, I think having a bill of materials that isn't tied to availability through the MaslowCNC's store front is a good idea.  Even if there's a primary link to the MaslowCNC's site with a secondary link to a comparable router bit on some other easily available site, maybe with a disclaimer, I think would be very beneficial.

People can decide for themselves whether they want to shop around for a cheaper or alternate router bit but providing a "known good" router bit helps in making that choice.  It's easier to tweak a known solution to an alternate solution than it is to create a solution from scratch.

abetusk, 2018-01-20:
Is this the actual bit that was used?

[image]

[0.25 in. x 1 in. Carbide Up Spiral Router Bit (2-flute, 0.25 in. chuck) ](https://www.homedepot.com/p/Diablo-1-4-in-x-1-in-Carbide-Up-Spiral-Router-Bit-DR75102/100660697)

If so, can we replace it in the Wiki?  There doesn't need to be a glowing recommendation to use this bit, but providing a link for a "known good" router bit goes a long way towards helping people trying to understand or assemble their own MaslowCNC.

BarbourSmith, 2018-01-22:
That looks like the right one. You are right that it's a good resource to keep, I agree with that reasoning

abetusk, 2018-01-23:
Wiki has been updated

closed 2018-01-23 by abetusk

BarbourSmith, 2018-01-23:
Excellent catch, excellent fix! 👍 👍 💯