MaslowCNC/Mechanics #96: Flexible frame Concept

CMHinckley, 2018-06-18:
While it seems that there several posts asking if there are ways to make the operation of the Maslow CNC work in a more vertical orientation, it seems that a better solution might be to mount the frame directly to the wall with hinges at the top. The bottom would be pulled outward (and braced in this position for machining) and when machining is finished, the frame is rotated flat against the wall. This could eliminate a lot of framing, saving money, fabrication time and significantly reducing the storage space when the unit is not in use. Has anyone worked on a concept such as this, and would anyone like to work on such a solution?

The precision of the cnc machines can be influenced by dimensional changes caused by moisture and temperature fluctuations, the flatness of the bed, and the stiffness of the frame. It might be possible to increase the stiffness and stability of the frame by using a grid of 1"x 4" between 2 external layers (one front and one on back) of 1/4" hardboard sheets, which are flatter and cheaper than plywood. This composite frame could be formed by gluing and screwing parts together, and the screws could then be completely removed after the glue has set, eliminating every possibility of hitting a screw during machining. A sacrificial sheet, could be used in front of the machine frame to avoid damage to the frame when machining. Has anyone worked on a composite frame of this type, and would anyone like to work on such a solution?

davidelang, 2018-06-18:
On Sun, 17 Jun 2018, CMHinckley wrote:
> While it seems that there several posts asking if there are ways to make the
> operation of the Maslow CNC work in a more vertical orientation, it seems that
> a better solution might be to mount the [...]

you don't actually save that much, just the rear legs and kickers.

There are a number of people who have done this, and it workswell.

> The precision of the cnc machines can be influenced by dimensional changes
> caused by moisture and temperature fluctuations, the flatness of the bed, and
> the stiffness of the frame. It might be [...]

actually, most of the frame makes very little difference in the machine accuracy 
(except if there is something that causes the sled to not slide, or the warpage 
is bad enough to affect the Z depth)

The machine accuracy is almost entirely defined by how rigid the top beam is, 
and how accurate the measurements during calibration are.

blurfl, 2018-06-18:
'''
mount the frame directly to the wall with hinges at the top.
'''
One of the beta testers, @Gero, built his fram this way, using a scissors jack to adjust the angle of the frame. It worked well for him.
'''
The precision of the cnc machines can be influenced by ... the flatness of the bed,
'''
I found this to be the case, the plywood I used for a workarea bowed and caused changes in the angle of the chain from the plabne of the sprocket. This would cause the chain to jump. It seemed to cause the sled to have additional frioction toward the sides of the workarea as well, but I don't have any data to prove it.

davidelang, 2018-06-18:
On Sun, 17 Jun 2018, Scott Smith wrote:

> '''
> The precision of the cnc machines can be influenced by ... the flatness of the bed,
> '''
> I found this to be the case, the plywood I used for a workarea bowed and caused changes in the angle of [...]

in your case this was inches worth of bowing to cause these problems

blurfl, 2018-06-18:
'''
in your case this was inches worth of bowing
'''
That's right. The composite frame that @CMHinckley suggests would be one way to prevent that kind of warping.

CMHinckley, 2018-06-18:
The max deflections of the top beam based on a total sled weight of approximately 17.5 lb (router, bricks, sled...) for the default frame is approximately 0.028" using Douglas Fir (Modulus of Elasticity of 1.8 million psi, and 2 x 4 framing). This includes the deflections resulting from the moment applied to the upright vertical members. This is a non-trivial portion of the target precision.  The best way to reduce the deflections is to increase the stiffness of the members and the bed.

blurfl, 2018-06-18:
That's true. Too, Doug fir is relatively strong but most users are using 'white-wood hem-fir' two-by lumber, which is less stiff and less stable.