MaslowCNC/Mechanics #97: Broken link or missing info

Shady-Grove, 2018-08-14:
https://github.com/MaslowCNC/Mechanics/wiki/0.25in-2-Flute,-Tungsten,-Up-Spiral-Bit

Instead of showing info on the 2 flute bit is returns me to home (parent menu).  Thanks!

BarbourSmith, 2018-08-14:
Very strange, this link works for me https://github.com/MaslowCNC/Mechanics/wiki/0.25in-2-Flute,-Carbide,-Up-Spiral-Bit

On which page are you clicking the broken link?

servant74, 2018-08-14:
Shady-Grove link uses the word Tungsten, where Bar's uses Carbide,

sounds like a little typeo onsome page.

BarbourSmith, 2018-08-20:
@Shady-Grove What page was it on so we can fix it and close the issue?

closed 2018-08-20 by BarbourSmith

Shady-Grove, 2018-08-20:
When I click on this link:

https://github.com/MaslowCNC/Mechanics/wiki/0.25in-2-Flute,-Tungsten,-Up-Spiral-Bit

It takes me here:

https://github.com/MaslowCNC/Mechanics/wiki

Thanks.

BarbourSmith, 2018-08-20:
Where are you seeing that link? The link takes you to a page which doesn't exist so you get bounced back to the front page of the wiki

Shady-Grove, 2018-08-27:
Sorry, your email went to the junk file I just found it.  The top link here:

https://github.com/MaslowCNC/Mechanics/wiki/Tested-Router-Bits-Guide

(0.250in 2-Flute, Tungsten, Up Spiral Bit<https://github.com/MaslowCNC/Mechanics/wiki/0.25in-2-Flute,-Tungsten,-Up-Spiral-Bit>)