MaslowCNC/Mechanics #98: Different hardware than assembly guide/ videos

PoliciaMalo, 2019-01-02:
I finally got around to assembling my maslow cnc and there are hardware differences between the site instructions and what I received.  

Is there an updated assembly guide for the new hardware?

Please...  I'm dying to finish assembly.

BarbourSmith, 2019-01-03:
Is the difference that you are missing the ring? 

If that is the case I recommend using this linkage kit: http://maslowcommunitygarden.org/Maslow-triangulation-linkage-kit.html and then you can follow the normal instructions substituting the linkage for the ring. It will be a little bit more work, but you will get much better results than without the ring

PoliciaMalo, 2019-01-03:
No I have the ring kit and z axis.  It's hardware that I ran into after
mounting the motors

BarbourSmith, 2019-01-03:
hmmmm I am not sure what hardware that would be. Can you post a picture of what part it is?

PoliciaMalo, 2019-01-03:
Ahh ok I had to go to the instructions for the original design and the
parts were chain guides they changed the nails as well as added bushings
for the string pulleys.

BarbourSmith, 2019-01-05:
Excellent!

Can I ask how you ended up with the out of date instructions? Maybe I can
fix something to make them less confusing in the future.

PoliciaMalo, 2019-01-05:
The instructions on the website under the new default frame do not mention
the parts used in the old default frame.  Specifically the use of nails
instead of screws in the center bungee bearings, the use of nails now on
the center bearings instead of screws.  The small 2 inch plastil bearing
used in chain tension for the old default frame.  Etc.  I'll elaborate more
in the dorms when build complete.

Also the bungees seem too long compard to the photos in the new default
frame build.  I just ordered new bungee spool in case needed.

Thanks.