MaslowCNC/Mechanics #100: Motor runs every time in same direction

mmmmmmmmmmmmmmm2, 2020-02-04:
Hi All please help me out, one of the motors is always running in the same direction, if i push up it goes up, if i push down it goes again up. and if i let it go to center it is every time somewhere else, and doesn't stop anywhere. what do i need to do...

JDogHerman, 2020-02-04:
Most likely the H-Bridge is dead on the controller card. 

You could eliminate other remote possibilities that cabling  by swapping the Left and Right motor wires running to the card.

BarbourSmith, 2020-02-04:
I think this sounds like an encoder issue to me. If the motor can't read the encoder value it will always spin in one direction. What do you see happen when you press Actions -> Test motors / encoders?

mmmmmmmmmmmmmmm2, 2020-02-05:
Hi All,
tested the motors, it says testing L motor direction 1 Fail , direction 2 Fail, and that for L / R and Z
See attached photo, it is brand new, must be something i´m doing wrong

[image]

BarbourSmith, 2020-02-05:
Did you close the issue because you found the solution?

mmmmmmmmmmmmmmm2, 2020-02-06:
No i didn't, still not solve it ..

Pushed wrong button

BarbourSmith, 2020-02-06:
Ok, let's figure out what is going on.

Do you see the motors move when running the test?

mmmmmmmmmmmmmmm2, 2020-02-07:
Hi all,

No, they do not move, when testing

But when i push manually it moves..

BarbourSmith, 2020-02-07:
Gotcha. When all three motors fail the test the issue is almost always with the power supply. Does the power supply have a blue light on it? Does the green light next to the plug come on when you plug the power supply into the controller board?

mmmmmmmmmmmmmmm2, 2020-02-07:
Good,

Correct, only two green lights.. no blue lights

BarbourSmith, 2020-02-07:
Where are the the green lights? Are they both on the controller board? Can you post a picture of the controller board and the power supply?

mmmmmmmmmmmmmmm2, 2020-02-07:
See attached photos

One of controllerboard and of the adapter

BarbourSmith, 2020-02-07:
I don't think the photos actually attached 😢

mmmmmmmmmmmmmmm2, 2020-02-08:
Mmmm, strange..

Will send them in 2 emails

mmmmmmmmmmmmmmm2, 2020-02-08:
Number 2

davidelang, 2020-02-08:
they did not come through, try adding them through the web interface.

David Lang

mmmmmmmmmmmmmmm2, 2020-02-09:
Okay, here they are
[image]

BarbourSmith, 2020-02-09:
I'm still not seeing the picture :( 

[image]

Let's move this conversation into the forums...I think we'll have less technical trouble there, you will get faster responses from more people, and that is really the right place for the conversation since this is about solving a particular issue instead of proposing a change to the design of the machine

BarbourSmith, 2020-02-09:
The forums are https://forums.maslowcnc.com/. You can login with your github login so no need to make an account or anything

mmmmmmmmmmmmmmm2, 2020-02-09:
[image]
First

mmmmmmmmmmmmmmm2, 2020-02-09:
[image]
Second photo

BarbourSmith, 2020-02-09:
Nice! Ok everything looks good there. And we are still seeing the motor move when you manually command them to move, but no movement at all during the test?

mmmmmmmmmmmmmmm2, 2020-02-09:
Yes, no movement during test, but can do manual but always same direction

BarbourSmith, 2020-02-09:
This is totally like anything I have seen before. The test should be overriding the controls in a way which makes the motor move if it has power...but the pictures and the fact that you can manually move it mean we have power

BarbourSmith, 2020-02-09:
Is the behavior the same on all the motors? The only thing I can think could be a bad connection in one of the connectors. Maybe let's try unplugging them all and re-plugging each motor very firmly

mmmmmmmmmmmmmmm2, 2020-02-09:
Strange isn't? Is an idea to set up software again?

BarbourSmith, 2020-02-09:
So strange 😲 

It could be worth setting up the software again, but this feels like a hardware issue to me

BarbourSmith, 2020-02-09:
I think checking the connections on the cables is the best place to start. I would unplug each motor and re-plug them firmly

mmmmmmmmmmmmmmm2, 2020-02-09:
It is new bought it end December , coming 2 weeks i'm not near my cnc, if you have any suggestions please let me know

BarbourSmith, 2020-02-09:
Did you try unplugging and re-plugging the cables to the motors? 

Have you contacted MakerMade? Do they have any suggestions?

mmmmmmmmmmmmmmm2, 2020-02-09:
Haven't tried to switch the motors but tried several times the connection. Have no idea who to contact assume there is some kind of warranty..

BarbourSmith, 2020-02-09:
They should send you a free replacement for anything which is not working...the tricky thing is I'm not even sure what to ask them for because I'm not sure what is going on. Maybe ask for a new arduino and controller board to be safe?

You can find them in the forums by posting your issue and tagging them as @makermadecnc they usually respond there pretty quickly