MaslowCNC/Mechanics #101: Bent Bolt

gamermom001, 2020-03-28:
I have one shoulder bolt that was bent when I received my order.  Can you send another one or send me the info I would need to order one?  Thanks

BarbourSmith, 2020-03-28:
Who did you buy your kit from? There are several sellers of the kit and I'm sure the right one can get you a replacement quickly

gamermom001, 2020-03-28:
My paypal account has Maker Made LLC listed.

BarbourSmith, 2020-03-28:
Awesome. You can contact them through:

https://www.makermadecnc.com/

Or find them as @MakerMadeCNC in the forums here: https://forums.maslowcnc.com

Basically I just do the designs, I don't actually sell anything so you will need to contact the people who sold the kit to get a replacement 😃

closed 2020-03-28 by BarbourSmith