MaslowCNC/Mechanics #102: Initial plunge off path

DJscreen, 2020-06-11:
We are cutting designs into 6x4 sheets of 6mm ply. The designs are 6" x 6" max and the cutouts are removed to create a lattice effect. There are two of the designs where the initial plunge through the material is about 2 mm off leaving noticeable "dings". We intend to experiment with 2 mm tabs at the relevant places and by inserting bulges into the svgs which will be manually finished later. 

The panel svg was set up using Inkscape v9 and the nc file generated in Krabzcam which as far as I can see is an excellent replacement for Makercam

BarbourSmith, 2020-06-12:
Hmmmm my first instinct is that this could be play in the router. Which type of z-axis are you using?

Another possible option is to add a lead in to your cuts which makes it so that the router bit doesn't raise and lower right along the edge of the part like this:

[image]

I am not familiar with Krabzcam, but I know CamBam supports it and they have a free trial version that might be worth playing with

DJscreen, 2020-06-20:
Thanks for your suggestions. We may have solved it by plunging the bit half way through the ply, withdrawing to safe height and then plunging to final depth. We have been using a 6mm twin flute cutter and maybe it is too stressed plunging the full depth in one go.

BarbourSmith, 2020-06-20:
That makes total sense. Nice problem solving!