MaslowCNC/Mechanics #103: Chain pull on setup.

alaskanalain, 2021-03-11:
On setup, the "measure chain" step kept pulling the chain until it stretched, the arms bent a lot and the motor eventually started to pull loose, if I didn't stop it I assume the motor would have burned out or torn loose. I beefed up the machine and now it's built like a tank.  
With the other chain, as before, I manually pulled the chain so there was a bit of slack then started to measure again, following instructions exactly, but it still bent the arms. It stopped and went slack as it should but not in time, so now the other chain is stretched too.  Both chains now skip on the teeth every time I try to do the test cut.  One tooth is slightly chipped too as a result (not too bad).  I have chain guides but it still slips due to the stretch.  The motors simply did not turn off and slack when they should.  The many youtube videos I watched have confirmed this.

I am reasonably good with machines but this has me near tears after many hours.  **Can I please get some new chains**, this time I will just use a tape measure, the measurements from the machine pulling them tight didn't yield accurate results either.

I went through the setup many times since and looked at this from every angle and don't think it's something I did wrong.  I have a small CNC and had no trouble with that at all.  Bicycle chains stretch and those are thicker than this, I assume that is why others also sometimes have issues with skipping.
Please help, Thank you
Alain

BarbourSmith, 2021-03-11:
I couldn't agree more that the calibration process is confusing and very much needs to be improved.

Which version of the kit are you using (ie who did you buy it from)? The chains and gears should be strong enough to handle the forces involved easily so I would be somewhat suspect of the build quality.

Which software are you using for calibration? WebControl or GroundControl?

alaskanalain, 2021-03-11:
Ground control and I think the seller was "Maker Made".  I got the kit with Z control, not sure what model, etc. It's one of the newer ones probably, got it a few months ago.  
It's well made but the motors didn't want to stop pulling, they did eventually but I don't think it should pull that hard, definitely straining every component, I could hear the creaking and motor strain sounds.  
I don't think the chains are made for that, far less beefy than my chainsaw and yet the chainsaw chain stretches on first use, it's designed with that in mind, Bicycles too.
Thanks for that, any ideas?  Do I have to buy new chains? Can I just measure the machine?  I am not even sure where the test cut should be on the plywood.

BarbourSmith, 2021-03-12:
I believe that they have their own software that might work better for you than Ground Control. I believe it's called Makerverse? Ground is pretty out of date at this point. 

You can manually measure, but it isn't usually accurate enough. 

Maker Made (https://makermade.com/) would be the people to get in touch with if you want some new chains.

BarbourSmith, 2021-03-13:
Of course, although I don't know if I really helped much :)

For better help come hang out in the forums:
https://forums.maslowcnc.com/

You can probably find someone with a similar experience there who might have some insights.

closed 2021-03-13 by BarbourSmith